larpix/larpix-control #7: Write getters/setters for configuration fields

peter-madigan, 2017-10-16:
Add getters and setters for Configuration class to prevent users from issuing bad commands.

samkohn, 2017-10-18:
Excellent idea. Could you write the relevant methods? I'm a huge fan of the `@property` and `@property.setter` features which let you call getters and setters as if they are just attributes (instead of `x = conf.get_foo()` you say `x = conf.foo` and instead of `conf.set_foo(y)` you say `conf.foo = y`). I used them for the Packet attributes, and it would be great to use them for Configuration as well.

I can write a bunch of test cases for them to defend against stupid/malicious users.

peter-madigan, 2017-10-18:
Will do.

samkohn, 2017-10-20:
Fixed by commit fa740d50832ecfc9fa71958ffbdc9b9495ccc700 (PR #8)

closed 2017-10-20 by samkohn