larpix/larpix-control #88: Proposal: new repo for scripts/tasks

samkohn, 2017-12-27:
I have been debating in my head for a while how to handle the always-evolving scripts and tasks that we have written for LArPix, while at the same time keeping the control library simple and clean.

I think the best way forward for larpix-control is to keep the basic/fundamental control library separate from tasks like threshold scans, noise tests, etc. And by "separate" I mean in a separate repository.

The files to move would be:

- larpix/tasks.py
- larpix/benchtest.py
- larpix/noise_tests.py
- larpix/analyzers.py (probably)
- scripts/internal_pulser_scan.py

This would entail a few changes:

1. (Obviously) the creation of a new repo, probably larpix-scripts or larpix-tasks
2. Stable and clear version numbering for larpix-control.
    - This almost exists already. I will soon christen v1.0.0. After that, the rightmost (aka bugfix) version number increments for bugfixes, the middle (aka minor) version number increments for feature additions that maintain backwards compatibility, and the leftmost (aka major) version number increments for new features that break backwards compatibility.
    - I will add a package-level attribute ``larpix.__version__`` so that scripts can inspect the current version. I will also add some functions to make it easy for scripts to check compatibility with the currently loaded version of larpix-control.
3. Resolution of a few installation bugs to make it easy to ``import larpix`` from another directory (which Dan was having trouble with).

@dadwyer @peter-madigan Let me know if you approve of this and I will begin this effort.

dadwyer, 2017-12-27:
Hi Sam,

I think this is a fine idea, but I would suggest we postpone this restructuring until after the first TPC demonstration.  I think the current software works well enough to achieve the TPC demo, and I am certain we will have plenty of fires to keep us busy between now and the DUNE collab meeting.  

After the TPC demo, we should have more time (and more experience) with which to consider a substantial rework of the code.

Regards,
Dan

samkohn, 2017-12-27:
OK

samkohn, 2018-06-15:
This was completed a few weeks (months?) ago

closed 2018-06-15 by samkohn