larpix/larpix-control #116: Packet structure decision

samkohn, 2019-05-31:
As we begin to associate additional data such as "chip key" to individual packets, I want to flag a semantic/structural distinction between the actual packet bits (and their interpreted values) and these additional pieces of data. My preference would be to keep these 2 concepts separate in the actual code. So for example:

```python
class LArPixDataPoint:
    def __init__(self):
        self.packet = None  # would be assigned to a Packet object
        self.chip_key = None
```

This way the Packet class handles only the UART bits.

Agree/disagree?

peter-madigan, 2019-05-31:
I think this is a good idea. Although I'm not sure about the naming.
Presumably these will also be used for configuration messages as well,
right? Maybe something like Message, Communication, Telegraph, etc fits a
little better.

samkohn, 2019-06-01:
Agree - and this contradicts #112 which wanted the global timestamp treated on the same level as ADC counts. So I will close that issue.

Really what makes the most sense to me is that this more-general structure should be called "packet" and the UART bits should be called "data bits" or "UART bits" or "ASIC bits" or something like that. But it's too late I think to rename the Packet class.

We really want this structure's name to indicate that it's all the data associated with a particular message. Maybe that's Message.

samkohn, 2019-06-25:
This seems to be more work than makes sense for right now. I will close this issue for the time being.

closed 2019-06-25 by samkohn