larpix/larpix-control #118: Tag new release for PyPI

peter-madigan, 2019-05-31:
The last release corresponds to v0.6.0, so we either need to update the readme or push some more recent releases to pypi.

larpixdaq, 2019-05-31:
I will push v2.0.0 to PyPi this evening

larpixdaq, 2019-05-31:
This is Sam by the way - apparently forgot to log out my phone from our larpixdaq account for the local DAQ - I'm about to get on a plane so I will neaten this up when I land.

peter-madigan, 2019-05-31:
Can you push 2.2.0? That is the interface that I would like the other
groups to use.

samkohn, 2019-05-31:
Yes. Will let you know as soon as 8t is complete. Do you want me to try to set up a script to automatically publish new tagged releases?

samkohn, 2019-06-01:
2.2.0 still has at least 1 major problem: it does not include the config JSON files because of the bug in setup.py. So I think a 2.2.1 is necessary. Right now I'm doing it as a branch off 2.2.0 but I'm open to your suggestions as to how to handle.

peter-madigan, 2019-06-01:
Oh weird. Didn't that get fixed earlier?

samkohn, 2019-06-01:
It did but only on a branch that wasn't merged back in

peter-madigan, 2019-06-01:
Ah ok. Yeah I guess tagging a 2.2.1 is the right thing to do.

samkohn, 2019-06-01:
Yeah the awkward thing is that we'll have a tag not on master but I think that's unavoidable

samkohn, 2019-06-01:
We can try it both ways. The problem with cloning the repo is you'd have to checkout a specific branch or tag since the head of master is not the right commit to work with.

samkohn, 2019-06-01:
One more thing is a license, now that we're going to be having many people looking at the code and potentially modifying it. I'm ideologically in favor of the GPL but pragmatically speaking whatever you want I'd be alright with (Public domain? MIT?)

samkohn, 2019-06-01:
For you to try out, v2.2.1-rc1:

https://github.com/larpix/larpix-control/tree/pypi-2.2.1

https://pypi.org/project/larpix-control/2.2.1rc1/

peter-madigan, 2019-06-01:
I'm all for the most permissive license possible.

samkohn, 2019-06-01:
In order of decreasing permissiveness (and somewhat simplified):

- [Public Domain](https://choosealicense.com/licenses/unlicense/) which means anyone could rebrand and license an exact copy and/or any modifications however they want
- [MIT](https://choosealicense.com/licenses/mit/) which means people must use the MIT license for exact copies, but any modified versions could use any other license
- Then there's the [LGPL and GPL](https://choosealicense.com/licenses/) which require all modified and derived versions to include the same license

peter-madigan, 2019-06-03:
I get this error when installing in a clean conda environment (python 2.7):
```
$ pip install larpix-control==2.2.1rc1
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting larpix-control==2.2.1rc1
  Using cached https://files.pythonhosted.org/packages/dd/03/8056e05b51232cd2f062d3926df6934e9600e3cd153423a8d227ff3e0c52/larpix-control-2.2.1rc1.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/63/4d8hyd3924b59nvqqbq89scm0000gn/T/pip-install-JtEfdD/larpix-control/setup.py", line 15, in <module>
        with open(os.path.join(here, 'VERSION')) as f:
      File "/Users/pmadigan/anaconda3/envs/py2.7/lib/python2.7/codecs.py", line 898, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/private/var/folders/63/4d8hyd3924b59nvqqbq89scm0000gn/T/pip-install-JtEfdD/larpix-control/VERSION'
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/63/4d8hyd3924b59nvqqbq89scm0000gn/T/pip-install-JtEfdD/larpix-control/
```
version 0.6.0 installs ok

samkohn, 2019-06-03:
Probably need to update MANIFEST.in to include VERSION

peter-madigan, 2019-06-03:
Added. Do you know when we can expect the package to be updated on pypi?

peter-madigan, 2019-06-04:
I've closed this since there is now a "temporary" tag to hold us over until the other branch is merged in.

closed 2019-06-04 by peter-madigan