larpix/larpix-control #119: Packet parsing issue with python2

peter-madigan, 2019-06-03:
From the DUNE slack channel:
Leon Mualem
I tried to get our setup running on a different computer so we weren't tied to Zoya's laptop.  Perhaps foolishly I went ahead and picked the latest bits from the github repository.  I did it as a download and unzipped it on our system.  Then did a pip install . from the new directory of goodies.  I think that all seemed fine, and was able to ping the daq board successfully.  (Once I fixed a typo in the IP address, another hour of my life gone for one missing 0.) It looks like transition to the different chip_key and chip_id is underway, so I tried to translate the things I did with Zoya.  I had some success, but got stuck, and I'm not sure if it's my translation, or if something else isn't quite right. (The former is the prime suspect.)  I'm not sure, since I get to a funny error at the end that looks like it is translating something in the config, but fails.  Maybe if there a code snippet to get the board configured and quiet, it will provide a pattern for the other commands as well.  This is about as far as I got.  Unfortunately this is just my code, and not the output and error.  I won't be touching it until Monday, but wanted to get this off my chest.
```
import larpix.larpix as larpix
from larpix.io.zmq_io import ZMQ_IO
controller = larpix.Controller()
controller.io = ZMQ_IO('tcp://192.168.100.137')
controller.load('controller/pcb-3_chip_info.json')
controller.io.ping()

#This command hangs:
#for chip in controller.chips: chip.config.load('chip/quiet.json')
#Try this: 

for chip in controller.chips.values(): chip.config.load('chip/quiet.json')

for chip in controller.chips.values(): print chip.config

for chip in controller.chips.values(): controller.write_configuration(chip.chip_key)
```
**The error:**
>>> for chip in controller.chips.values(): controller.write_configuration(chip.chip_key)
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "larpix/larpix.py", line 1158, in write_configuration
    self.send(packets)
  File "larpix/larpix.py", line 1068, in send
    self.io.send(packets)
  File "larpix/io/zmq_io.py", line 36, in send
    msg_datas = self.encode(packets)
  File "larpix/io/zmq_io.py", line 123, in encode
    msg_data = [b'0x00%s 0' % packet.bytes()[::-1].hex().encode() for
packet in packets]
AttributeError: 'str' object has no attribute 'hex'
```

Follow up from Lane Kashur:
I also received "AttributeError: 'str' object has no attribute 'hex'" initially when using python 2.  Using the same code in python 3 seemed to solve that issue for me

samkohn, 2019-06-03:
> File "", line 1, in

We should add a unit test to catch this error and then also fix it.

peter-madigan, 2019-06-03:
Minimal code to raise error:
```
from larpix.larpix import Packet
from larpix.io.zmq_io import ZMQ_IO
ZMQ_IO.encode([Packet()])
```

closed 2019-06-03 by peter-madigan