larpix/larpix-control #147: chip_key thoughts from Dan

samkohn, 2019-06-20:
A conversation with Dan about data format led to some discussion of chip_key

The gist is that it would suffice to uniquely identify ASICs by a generic "network ID" + the encoded chipid from the ``Packet`` object properties. The network ID would just be a number that maps to a particular IO chain or Hydra I/O network (or some other semantics that's interpretable by the IO controller).

peter-madigan, 2019-06-21:
OK, I think that's reasonable. That way it doesn't require some unknown length string.
I'll have to give some thought to the configuration file that we should use to set that up. Probably will need a new zmq io configuration that maps these chip keys to IP addresses and io channels.

samkohn, 2019-06-25:
Recording our conversation today with Dan:

We will use 3 fields - DAQ board ID, IO channel ID, and chip ID - to identify each ASIC. This will involve changes mostly to the ``Controller``, ``IO`` and ``Logger`` classes.

closed 2019-07-03 by peter-madigan