larpix/larpix-control #165: test_tutorial.py takes a long time on the second test

samkohn, 2019-07-02:


peter-madigan, 2019-07-03:
I think this might be because I am using a valid address for the ZMQ_IO class initialization - my guess is that there is some timeout period where it is trying to establish a connection. This fix is likely to use a designated "bad" address, e.g. 192.0.2.1 like what is used in the zmq_io tests.

samkohn, 2019-07-03:
After much hunting, I have determined:

- The delay is only present when connecting to an address ending in ``.local``
- The delay is due to waiting for a background thread to finish executing. The thread starts during the ``socket.connect()`` call. I believe this is the bonjour / avahi networking driver timing out when it cannot find a connection.
- The delay stacks up if there are multiple sockets connecting.
- I believe the fact that the delay is not interrupted during ``socket.close`` or ``context.term`` is likely a bug. I'm not sure whether it's in pyzmq or libzmq.

The solution is to not use ``.local`` addresses during the tests

PR #169 (2019-07-05): Fix networking bug in test_tutorial (ugh)
Fixes #165

ZMQ hangs in context.term if
1. the socket connects to an address ending in ".local" that does not
    exist
2. 5 seconds have not yet elapsed, per attempted socket connect

This seems to only be present in our test_tutorial.py script. I fixed it
by replacing the daq-srv1.json config call with a new loopback.json file
that just connects to localhost.
commit e8062ce7 (2019-07-05): Fix networking bug in test_tutorial (ugh)

Fixes #165

ZMQ hangs in context.term if
 1. the socket connects to an address ending in ".local" that does not
    exist
 2. 5 seconds have not yet elapsed, per attempted socket connect

This seems to only be present in our test_tutorial.py script. I fixed it
by replacing the daq-srv1.json config call with a new loopback.json file
that just connects to localhost.
commit eb71648f (2019-07-05): Merge pull request #169 from samkohn/issue/165-tutorial-hangs

Fix networking bug in test_tutorial (ugh)

Fixes #165
closed 2019-07-16 by peter-madigan