larpix/larpix-control #179: dataserver_message_decode is slow (bottleneck)

samkohn, 2019-07-25:
@peter-madigan ran into issues running on the v1.5 PCB when it's first started up and the ASICs are producing triggers at the maximum rate. The ZMQ_IO couldn't parse the incoming messages fast enough, and data was lost. The saturated packet rate is approximately 100kHz.

I ran a simple test to quantify the problem:

```python
In [3]: packets_0 = (Packet() for _ in range(100000))                                                                                                                                         

In [4]: packets_1 = []                                                                                                                                                                        

In [5]: for p in packets_0: 
   ...:     p.chip_key = '1-1-40' 
   ...:     packets_1.append(p) 
   ...:                                                                                                                                                                                       

In [6]: encoded = dataserver_message_encode(packets_1, key_parser=lambda key:dict(io_chain=key.io_channel))                                                                                   

In [7]: %timeit dataserver_message_decode(encoded, key_generator=lambda chip_id, io_chain, io_group:Key(io_group, io_chain, chip_id), io_group=1)                                             
1.01 s ± 11.1 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

Clearly it is not good if it takes 1 second to parse 1 second's worth of data. I will profile the code and try to speed it up to 0.5 seconds to parse 100000 messages.

peter-madigan, 2019-07-25:
Sounds good. Just for reference, the plan is to have 16 channels / daq board. Each DAQ board will have ~100*64 pixels, so a standard rate for Argoncube could be up to ~100kHz.

samkohn, 2019-07-25:
Significant progress:

```python
In [3]: packets_0 = (Packet() for _ in range(100000))                                                                                                                                         

In [4]: packets_1 = []                                                                                                                                                                        

In [5]: for p in packets_0: 
   ...:     p.chip_key = '1-1-40' 
   ...:     packets_1.append(p) 
   ...:                                                                                                                                                                                       

In [6]: encoded = dataserver_message_encode(packets_1, key_parser=lambda key:dict(io_chain=key.io_channel))                                                                                   

In [7]: %timeit dataserver_message_decode(encoded, key_generator=lambda chip_id, io_chain, io_group:Key(io_group, io_chain, chip_id), io_group=1)                                             
682 ms ± 4.68 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

Changes:
- ``Packet.ship_key`` (setter): only create a new Key object if the supplied value is not already a Key [-170ms]
- ``Packet.chip_key`` (setter): only assign to self.chipid if the key's chip_id is different [-100ms]
- ``MultiZMQ_IO.generate_chip_key``: final Key is now created using the 3-argument constructor rather than using ``from_dict`` [-80ms]
- ``Packet.__init__``: don't use ``bitarray.pop(0)`` twice in a row, which is inefficient. Instead take a slice of ``[2:]`` [-30ms]
- ``bitarrayhelper.fromuint``: Use faster ``zfill`` rather than concatenating strings. [-10ms]

total: -310ms, ~30%. The change from ``MultiZMQ_IO`` is not reflected in this total because it was not used as the ``key_generator`` in the "overall" timing test.

samkohn, 2019-09-12:
Turns out the Blizzard DAQ computer is slower than my laptop. So we need even more speed!

New idea:
1. Packet object has attributes ``_io_channel`` and ``_io_group``
2. These attributes are set directly rather than supplying a Key
3. The getter for Packet.chip_key will return an ephemeral Key object (i.e. a new one each time Packet.chip_key is called, rather than saving one as a private attribute)
4. The setter for Packet.chip_key will directly update the relevant attributes

This will remove 2 of the 1-us-per-packet calls (new Key and p.chipid) which should give another 25%-30% speedup. (fingers crossed)

[image]