larpix/larpix-control #180: Naming issues for PCB-5 / PCB-1.5

samkohn, 2019-08-05:
In various places in the LArPix documents/communications/hardware we refer to the prototype packaged-ASIC PCB as PCB-1.5 and PCB-5. I propose unifying around PCB-1.5.

peter-madigan, 2019-08-06:
I think these refer to two separate ideas. The v1.5 anode refers to a
particular design of the anode (four chips, chip IDs 3,5,10,12, single
board, external pulser, etc). PCB-5 is a label for a particular v1.5 anode.

samkohn, 2019-08-06:
OK - there's a config file, ``configs/controller/pcb-5_chip_info.json``, which I think should be renamed to ``pcb-1.5_chip_info.json``, since it refers to the PCB layout

peter-madigan, 2019-08-06:
No, it refers to pcb-5 which has a different daisy chain configuration from
the default v1.5 anode.

samkohn, 2019-08-13:
Recording our discussion:

- We will maintain a distinction between a specific physical PCB instance (each copy of a PCB) — which we will identify as PCB-1, PCB-2, etc. — and a pixel-to-(chip+channel) configuration (determined by the PCB layout design), which we will identify as layout-1, layout-2, etc.
- File names in ``larpix/configs/controller`` will refer to PCB numbers. File names in larpix-geometry will refer to layout numbers.
- There's a potential for the controller config (aka PCB) files to specify the appropriate larpix-geometry (aka layout) file.

peter-madigan, 2019-08-13:
Seems reasonable to me. This distinction is not case-sensitive right? I don't want to have to rename the `pcb-X` files to `PCB-X`.

samkohn, 2019-08-14:
Yeah the files will be pcb-1.json, pcb-2.json, etc. exactly how they
already are.

samkohn, 2019-08-15:
Something to consider for the future: in LArPix v2, the Chip ID will be configurable in software rather than hardware, which means we will need to introduce a new concept + term for the "location"/"socket" ID on the PCB (i.e. which physical location) that determines which pixels the ASIC connects to, as distinct from the Chip ID, which is now just a flexible "software address" rather than a fixed physical location.

Of course, we may (should?) decide on standardized Chip ID configurations for a given layout. But I think LArPix Control and LArPix Geometry should be able to smoothly handle changes to Chip IDs [e.g. by changing ``controller/pcb-XYZ.json``] without requiring an entirely new layout file.

closed 2019-09-04 by samkohn