larpix/larpix-control #181: verify configuration fails below a certain value of global threshold/trim threshold

zoyavallari, 2019-08-21:
For example for Chip 10:
If all channels are unmasked and trim threshold(TT)  = 31 for all channels
verify_configuration() returns (True, {}) until global threshold (GT) is 23.
For values of GT below, it returns a bunch of (False, {Key('1-1-3'): {0:16, None),.....)}}).
If I unmask only one channel at a time, then it's a different values for each channel showing correlation with data rate of that channel.
For instance, for Channel 5: GT = 22, TT =3 returns False for verify configuration
Channel 13: GT = 11, TT = 28 return False and so on.
This behavior is also present for other chip_keys.

peter-madigan, 2019-08-21:
At some point (~100kHz of packets), the data rate from the chip is saturated. At this point, there is no guarantee that a data packet will make it back from the chip.

My guess as to what is happening is that at this particular global threshold value, the pedestal voltage is greater than the discriminator threshold voltage and so you are saturating the data rate from the chip and thus filling up the chip FIFO. When you perform a verify_configuration, the larpix-control sends a read configuration packet from each register. The chip attempts to insert the response packet into the FIFO, but since the FIFO is full, this is unsuccessful. Thus you receive no packets back from the chip (hence the `None`). You could check this by looking at `controller.reads` (each chip checked will fall under a separate read entry).

zoyavallari, 2019-08-24:
Everything makes sense after Peter told me that coarse global threshold scan of configure_to_rate.py script is only run for 100ms. I was taking the values after the coarse scan (for 100ms run) and then running it assuming that it was the data rate at 1s, so I was in fact at ~100kHz of packets when I assumed I was far away from the saturation. Everything runs as expected.

closed 2019-08-24 by zoyavallari